eclipse-paho/paho.mqtt.embedded-c #218: ESP32-Mqtt was not connected using Azure SSL Certificate

selvamani1991, 2021-03-10:
Hi,

ESP32-Mqtt was not connected using Azure SSL Certificate

I am using the below code for connecting with ssl certificate was not working.

#include <WiFiClientSecure.h>
const char* ssid = "xxx"; // your network SSID (name of wifi network)
const char* password = "xxx"; // your network password
const char* server = "xxx.azure-devices.net"; // Server URL
const char* test_root_ca=
"-----BEGIN CERTIFICATE-----\n"

"-----END CERTIFICATE-----\n";

// You can use x.509 client certificates if you want
const char* test_client_key =
"-----BEGIN RSA PRIVATE KEY-----\n"

"-----END RSA PRIVATE KEY-----\n"; //to verify the client

const char* test_client_cert =
"-----BEGIN CERTIFICATE-----\n"

"-----END CERTIFICATE-----\n"; //to verify the client
WiFiClientSecure client;
void setup() {
//Initialize serial and wait for port to open:
Serial.begin(115200);
delay(100);
Serial.print("Attempting to connect to SSID: ");
Serial.println(ssid);
WiFi.begin(ssid, password);
// attempt to connect to Wifi network:
while (WiFi.status() != WL_CONNECTED) {
Serial.print(".");
// wait 1 second for re-trying
delay(1000);
}
Serial.print("Connected to ");
Serial.println(ssid);
client.setCACert(test_root_ca);
client.setCertificate(test_client_cert); // for client verification
client.setPrivateKey(test_client_key); // for client verification
Serial.println("\nStarting connection to server...");
if (!client.connect(server, 8883)){
Serial.println("Connection failed!");
}
else {
Serial.println("Connected to server!");
}
}
void loop() {
// do nothing
}

scaprile, 2021-03-10:
This is not the proper repo.
`WiFiClientSecure` does not belong here:
```
[paho.mqtt.embedded-c]$ grep -R WiFiClientSecure *
[paho.mqtt.embedded-c]$
```
Go to the place where you got your code from and ask there.
You might be having WiFi or TLS issues, or you might not have properly configured your device with your provider; make sure you report _what_ does not work.

selvamani1991, 2021-03-10:
Here i attched the screenshot, I don't have any error, only server was not connected.
https://prnt.sc/10hzcyf

selvamani1991, 2021-03-11:
Hi, I am using this below code, it shows Failed message.

#include <WiFiClientSecure.h>
#include <PubSubClient.h>

const char *ssid     = "xxx";     // your network SSID (name of wifi network)
const char *password = "xxx"; // your network password

const char *server = "xxx.azure-devices.net";  // Server URL

const char* test_root_ca= \
"-----BEGIN CERTIFICATE-----\n" \
"MIIDxDCCAqwCCQDyrMM0eCtQEDANBgkqhkiG9w0BAQsFADCBozELMAkGA1UEBhMC\n" \
"SU4xCzAJBgNVBAgMAktBMRIwEAYDVQQHDAlCYW5nYWxvcmUxEzARBgNVBAoMCkJy\n" \
"YWluQ2hpbGQxGTAXBgNVBAsMEEF6dXJlU21hcnREZXZpY2UxFzAVBgNVBAMMDlNt\n" \
"YXJ0RGV2aWNlQmN0MSowKAYJKoZIhvcNAQkBFht2aWpheS5uQGJyYWluY2hpbGR0\n" \
"ZWNoLmNvbSAwHhcNMjAwNDEyMDYxNzM2WhcNMzEwNjMwMDYxNzM2WjCBozELMAkG\n" \
"A1UEBhMCSU4xCzAJBgNVBAgMAktBMRIwEAYDVQQHDAlCYW5nYWxvcmUxEzARBgNV\n" \
"BAoMCkJyYWluQ2hpbGQxGTAXBgNVBAsMEEF6dXJlU21hcnREZXZpY2UxFzAVBgNV\n" \
"BAMMDlNtYXJ0RGV2aWNlQmN0MSowKAYJKoZIhvcNAQkBFht2aWpheS5uQGJyYWlu\n" \
"Y2hpbGR0ZWNoLmNvbSAwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDM\n" \
"9BSby0MzYNe4FpezzhUwkthQl6a41zuf6bpp42jUbagGjmLvBf36By6dUd4qQN9n\n" \
"DELU/51TLr0v3lWRyli9dz9gaD3l0l8jw7mT5TaaxgWi6A8o288T8Vlf2QKPwH8Y\n" \
"4wejrVpWp4hUxX5SA6y5a4XTe1UBHyKDiAAa3SGUu7gzw6zau+6HE45sQhzBZoke\n" \
"fApDQvYmexjecIfqGpNTNxhvKENTt/luNBRevsRr5uEp+7D2mwVHWQxz321EvtJX\n" \
"8GePNpzpavBzfQaXBIg8N8Ump33NEzPcfCvlVjY/GCTdjDV47RvFCB45hgoU5y1P\n" \
"voSEu60ju4fOnKK5z9OlAgMBAAEwDQYJKoZIhvcNAQELBQADggEBAAqrBglSslip\n" \
"kmrilWXTRWwfc9goH+BbgHvSLJJI6/LEsLVkHPYE+Id4lvyFih8JMTgn5QVqDdtK\n" \
"dqE8MN792VQBmDMMWamRuQLWAjZi3zrRlT24X1S0lQncx418z4gIJL7rugfdkkdc\n" \
"v77BXQaog3N7ErEo0ye8AajgcnNixe5oYzI0jrC/7TFpo6r6me5yKz2GLq/RIpSq\n" \
"8pT2prkw0pCJWBHqT7pt/f2ZxPp5wKFKqOXdWJSW957hDcVobgYUzfARtBA1pgmv\n" \
"JdA4Wz2GMmzdhKGlNKG0wXjesSYSyyijKfNHQzRZ04SM4kDk5Xf9sm1igWvrOIbU\n" \
"B6qOPjRQhDA=\n" \
"-----END CERTIFICATE-----\n";

// You can use x.509 client certificates if you want
const char* test_client_key =  \
"-----BEGIN RSA PRIVATE KEY-----\n" \
"MIIEpAIBAAKCAQEAvrsh1YjvyGB90UroWLX383OrtrfS5jabzIAl99W0gjQWpN8E\n" \
"Ccq8thffWzq6+EoJrjrMshqLrgwuupsKyYAxZ8pEBMu/7/6P2ROk/9huycP9MP/c\n" \
"nexpfqlTrlrabDFG4YEbFNGmQ0eFhGjR4G+ymn6WGGzcG6c+Wd7YgseMI7MPz8W0\n" \
"EUXSmQMYa0BIrFfwQ4j5cnJL2L6ylenc6rwoVG1Zzlvx+1ZgtOuIMjZDzyfwD2OU\n" \
"grB8Spjg8RI8CPoBeF8TMwnNghPfSJZ2Hl7Kq6S8Wxo67qfUVBVtfA8lE/8dvO16\n" \
"txBxc7xDnZDJD6bQP+qZl81wLjzu6/e2H6Cd7wIDAQABAoIBAFFbW1RNhBKR7jk/\n" \
"XZqxjCGBnqZjBQgEFTMHGqr7snKzq1EKfBt9Kj/2MOUmSSit1DFIjZYPMrO8KJaa\n" \
"meBX2TsvkRIGo0nI6S0sbv5fIOrZuf+TtFR8yocnzTUbNf5eiqCYRKf9wZEDShWv\n" \
"7I5yttml03p5vO3iyfrlzq8CVWnIM03lDgjgVRvgqPMTL/ZpOCkbSuMGPlwTVXXR\n" \
"CTBH2SNg3SRROKMYM4sJWGbbCeB55VkryIosfpSkqeOxJ2Kon5j9PNOt+3M80/pV\n" \
"T/t4gRtp9IlFu2AfGbibSiyBs2/Zx0VPJUlPvJX57k88EtuCZz/ZOxc3ryRKrJN1\n" \
"wMg70ikCgYEA8l4m82UFivtPkE7fcmNtLUTilwh9v5Fkl9HTcJx0uppa47+wnJHU\n" \
"+s1uBh9O1yjaxJS9bBupNsNZc7/93SP6e6m2/mBP8AR3SmmY7bJTw2hnN8caMSVW\n" \
"tm57NY6kLNF5BU6bZiBQ6YqFExOBJ4PGp8LOhYDnQ6NuA9zcVZjvxW0CgYEAyXV2\n" \
"iIA8hnjAqkxq61B2EhXxUigCqAFDQGJHt98F92it+GeRU5iWBjSWwrvPuiCMSRNl\n" \
"bPKB0gzPCjag8FGK9B0hHApCYpgQuZ5x80P0txkWYP1dYB2+zsKCr32WMaw8pDPD\n" \
"ZmOoaLIJVcNpVCHPaUIHnvpwfvE6cEE0fCLng0sCgYEA487bjatGqnvNLTb1j+8k\n" \
"j4hP8uZoS8e9ZXYz54Unsr2i8K8csg3gwlyVLev/jv03o86qvRLC2laDnei4FUPc\n" \
"DKBZAIBGorOP7MGUnNchJGrdme/rDoaqGo99/7kzV20M85APQM7fPqSUsHmEzD9w\n" \
"Wf8hXxrqRQk4Pyk5cLtvo4ECgYB7V9Sb3E+yUSo82g64RgoUJqUalO6TaDqBp+qb\n" \
"FAw2Hzwsk1Rx50ErSXVogs66+YSKlTqLCqsbLMZyTWIZ4QL/VPpZjZp5fPr4ep7v\n" \
"vxKMK+r8tQM/fta9BnA2++BCR17Vde/Tt357o0192eeaIB06J7g8yL6pqg4KfpGg\n" \
"00oFowKBgQCwCkR5b2OJqsYVcLmkivSUq+fvn++CZ25bL/fp2tnjyqKn2wmXJQb2\n" \
"zyOOvmQLeMibg2N0C5w58uVK29iSB4wvZR6ld22nwiNuMrgCqXSPfala9diShiT4\n" \
"jKR5LyiKNwbWUIhlHRoEoflU99R9phQ9RBCLM9DZPl14l0t121BrTA==\n" \
"-----END RSA PRIVATE KEY-----\n";   //to verify the client

const char* test_client_cert =  \
"-----BEGIN CERTIFICATE-----\n" \
"MIIDUDCCAjigAwIBAgIEDWcmzDANBgkqhkiG9w0BAQsFADCBozELMAkGA1UEBhMC\n" \
"SU4xCzAJBgNVBAgMAktBMRIwEAYDVQQHDAlCYW5nYWxvcmUxEzARBgNVBAoMCkJy\n" \
"YWluQ2hpbGQxGTAXBgNVBAsMEEF6dXJlU21hcnREZXZpY2UxFzAVBgNVBAMMDlNt\n" \
[... 75 lines cut ...]
[E][esp32-hal-cpu.c:170] setCpuFrequencyMhz(): Can not switch to 240 MH⸮⸮⸮⸮⸮CPU frequency rated for 160MHz.
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 2 = 160 Mhz, APB: 80000000 Hz
Attempting to connect to SSID: Libre
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
.[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
..[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.61, MASK: 255.255.255.0, GW: 192.168.1.1
Connected to Libre

Starting connection to server...
[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 280512
[V][ssl_client.cpp:65] start_ssl_client(): Starting socket
[V][ssl_client.cpp:104] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:113] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[V][ssl_client.cpp:197] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:212] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:233] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:242] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:257] start_ssl_client(): Free internal heap after TLS 238484
[V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 23 bytes...
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.

Failed.

CIPop, 2023-07-18:
> This is not the proper repo. `WiFiClientSecure` does not belong here:
>
> ```
> [paho.mqtt.embedded-c]$ grep -R WiFiClientSecure *
> [paho.mqtt.embedded-c]$
> ```
>
> Go to the place where you got your code [...]

I agree with @scaprile : since your TLS layer did not connect, Paho MQTT is not yet involved. 

My team maintains https://github.com/Azure/azure-sdk-for-c-arduino - please see our sample on how to properly set up TLS to connect to Azure. If you still have issues, please file an issue within our repo instead.

icraggs, 2023-11-29:
Closing as per the above comments.

closed 2023-11-29 by icraggs