eclipse-paho/paho.mqtt.embedded-c #219: MQTT auto disconnected after KeepAlive time

HoangPhuongPKH, 2021-03-11:
Hi all,
Sorry about my English,
I'm working with MQTT and STM32, FreeRTOS and lwIP
I followed this a example
https://github.com/eziya/STM32F4_HAL_ETH_MQTT_CLIENT
No problem if publish to STM32 topic subscribed every second.
But, if I connect to broker (mosquitto), then do nothing, after 60s keepalive, STM32 send PINGREQ and moquitto send back, but I debug that STM32 did not recevied PINGREQ and then disconnected.
In this example, if STM32 reconnect if disconnected, but at 4 times, it can't connected.
Anyone can help me, thanks!

scaprile, 2021-03-11:
You must able to determine where the broker response gets lost, "STM32" is not the name of a module, analyze your application and determine where you have the problem. Then
If you have a problem with an STM32 application, place an issue in that application repo, or go to an ST forum and ask there
If you have a problem with lwIP, go to the lwIP forum and ask there
If you have a problem specific to Paho, then ask here, please properly describe what you send, what you get, what you expect to get, and why.

Ivan-Bolshakov, 2021-06-10:
Hello, I have the same problem right now. If you can solve it? I will be very grateful for the solution to this problem

scaprile, 2021-06-10:
> Hello, I have the same problem right now. if you have solved the problem, I will be very grateful for the answer
Same answer: https://github.com/eclipse/paho.mqtt.embedded-c/issues/219#issuecomment-796828932

icraggs, 2023-11-29:
I think we are adressing the issue (see the referenced commit above).

closed 2023-11-29 by icraggs

haneul49, 2024-09-05:
I think this might be helpful for you.
https://github.com/eclipse/paho.mqtt.embedded-c/issues/219#issuecomment-796828932

z00kiii, 2025-02-24:
https://github.com/eclipse-paho/paho.mqtt.embedded-c/issues/122#issuecomment-362876332

solved this issue for me