eclipse-paho/paho.mqtt.embedded-c #224: Performance problem related to cycle() and MQTT_TASK

tmaltesen, 2021-07-26:
I have need to publish several times each second.

I have found that the design of MQTTRun and cycle() design does not work very well. First the MQTTRun will block transmission so that it is not possible to transmit more than two messages pr. second.

This is because every time the MQTTPublish will release the mutex on cycle(), the MQTTRun thread will get a chance to run and will block waiting to receive for 500 milliseconds.

Changing the receive timeout to something less than 500 milliseconds does not work well either, my RTOS/network will simply have a minimum value for receive.

I have tried to modify the MQTTRun so that it is waiting on an idle timer before running so that main wait time is spend in MQTTPublish. Byt this give me problem with dropping connection when receiving messages instead, I have not yet found the reason, but I expect it to be related to error handling in the cycle() function.

tdk2nyt, 2022-05-10:
@tmaltesen  I meet the same problem, I tried to use select to wait incoming message, but there were packets lost and reconnection. Is there any good idea? Thanks!

tmaltesen, 2022-05-10:
I have made some modifications to the source code delivered by Keil. But it is still not perfect, especially the way cycle() is used to process message are really stupid. If the transmit code is waiting for a respons any incomming messages are lost... :-(
My version is running better than original but not really good.
I am planning to rewrite the module with OS tasks, queues and a state machine, but have not done it yet.
Please check you heap, at an early point i missed my heap was too low....