eclipse-paho/paho.mqtt.embedded-c #226: Is there a limit on the amount of bytes the payload must be in the publish message?

hakam00ra, 2021-09-27:
Anything over 600 bytes malforms the packet and it cannot be published.

I can't see any apparent limit in the code or the MQTT standard.

thanks

scaprile, 2021-09-27:
MQTTSerialize_publish() just calls memcpy(), it does not modify your payload; [see source code](https://github.com/eclipse/paho.mqtt.embedded-c/blob/29ab2aa29c5e47794284376d7f8386cfd54c3eed/MQTTPacket/src/MQTTSerializePublish.c#L83).
Is it the header you see is not OK ? How ? What do you do, how do you call this function, what do you see, do you have actually have enough free memory where your pointer points to ? What have you checked, have you stepped the code ?
MQTT msg length can be encoded in with a different number of bytes and handle up to 256MB; the actual system memory limits will mostly come first.

hakam00ra, 2021-09-27:
This is how i call the function
```
unsigned char bufm[1500];
int bufmlen = sizeof(bufm);
rc = MQTTSerialize_publish(bufm, sizeof(bufm), dup, qos, retained, msgid, topicString[0], message, messageLen);
```
I get this in my console for a particular payload: 
```
messageLen=580
sizeof bufm=1500
rc=600
```
The payload (bufm) is transmitted fine.

For a different payload:
```
messageLen=638 
sizeof bufm=1500
rc=658
```
And since the payload is larger than 600 bytes the payload (bufm) is now only 19 bytes and only contains some parts of the message array.
Notice how the rc is correct on both occasions.

Also there is enough memory where my pointer points to `unsigned char bufm[1500];`

thanks

scaprile, 2021-09-27:
Make sure your topicString[0] is a [MQTTString](https://github.com/eclipse/paho.mqtt.embedded-c/blob/29ab2aa29c5e47794284376d7f8386cfd54c3eed/MQTTPacket/src/MQTTPacket.h#L81)
Execute your code step by step and pinpoint where your payload is trashed.
Remember memcpy() belongs to your system.

hakam00ra, 2021-09-28:
yup.....
It was a stack issue. My pointer was pointing to the stack...
Sorry and thanks for the help.

scaprile, 2021-09-28:
Please close the issue, the moderator is busy

scaprile, 2021-09-28:
> Execute your code step by step and pinpoint where your payload is trashed.
> Remember memcpy() belongs to your system.

If you do find something in the code, I'll be happy to help (maybe others too)

closed 2021-09-28 by hakam00ra