eclipse-paho/paho.mqtt.embedded-c #232: Does this library support mqtt v5?

Zapnologica, 2021-12-09:
We are looking to use the ResponseTopic feature outlined in v5.0 MQTT protocol>

Does this library support it, and if not are there plans to support it and when?

scaprile, 2021-12-09:
Possible duplicate of issue #102

KonssnoK, 2022-05-12:
the answer is no.
Looking partially at implementation there is no support for properties in the publish serializer nor the AUTH packet.

lmapii, 2022-05-20:
I just see that the last commit is from 5 years ago, is this project still active?

CIPop, 2023-07-18:
ResponseTopic should work as expected for code in the `develop` branch. I have added samples that use properties: https://github.com/eclipse/paho.mqtt.embedded-c/blob/develop/MQTTPacket/samples/v5pub0sub1.c#L221

> is this project still active?

Yes, it is! I am actively working together with @icraggs on wrapping up MQTTv5 support that was already available for MQTTPacket in the `develop` branch. I am a developer with Microsoft Azure IoT and my work is sponsored by our group.
Next on my list is to create a `MQTTClientv5-c` equivalent.

icraggs, 2023-11-29:
I'm going to close this, as we are working on a release with MQTT 5 support.

closed 2023-11-29 by icraggs

Singa17, 2024-05-08:
> ResponseTopic should work as expected for code in the `develop` branch. I have added samples that use properties: [...]

[CIPop](https://github.com/CIPop)
Hi CIPop, Any update on the MQTTClientv5-c?