eclipse-paho/paho.mqtt.embedded-c #235: Client never closes socket

pavel-kamaev, 2022-02-05:
c->disconnect() should be called inside MQTTCloseSession()

scaprile, 2022-02-05:
What is 'c' ? In which module ? In which file ?
There is a MQTTCloseSession() function in MQTTClient.c in MQTTClient-C that has a parameter named 'c', which is a MQTTClient structure; there is no member named 'disconnect' in that structure. Are you using this very repo or a forked/vendor modified one?
The client does not know what a socket is, it is network agnostic, whatever you are using as a transport support has to handle the TCP connection to the broker, including opening and closing the socket (if you use a socket API) and handling network errors.
For example:
- the Linux port has NetworkDisconnect() which closes the socket, as you can see on the sample _MQTTClient-C/samples/linux/stdoutsub.c_.
- the FreeRTOS port has a 'disconnect' member in a 'Network' structure that handles that.

pavel-kamaev, 2022-02-05:
I use this repo with FreeRTOS and LwIP stack.

I do something like that:
```c
while (1) {
  do {
    NetworkInit(&network);
    MQTTClientInit(&client, &network, 30000, sendbuf, sizeof(sendbuf), readbuf, sizeof(readbuf));
    if ((rc = NetworkConnect(&network, broker_addr, 1883)) != 0) {
      continue;
    }
// Fill connectData
    if ((rc = MQTTConnect(&client, &connectData)) != 0) {
      continue;
    }
// Subscribe
  } while (rc != SUCCESS);
  while (1) {
    if ((rc = MQTTYield(&client, 500)) != 0) {
      break;
    }
  }
}
```

What is the proper way to handle connection errors (reconnect to broker on error)? Should I close connection every time MQTTYield() returns negative value or FAILURE only? Is there any reason not to use disconnect callback from MQTTCloseSession()?