eclipse-paho/paho.mqtt.embedded-c #251: PlatformIO's last release outdate

Alain-Godo, 2023-09-12:
I noticed that the last version of the lib is not updated in PlatformIO register:

[image]

CIPop, 2023-09-19:
Direct link, for reference: https://registry.platformio.org/libraries/eclipse/Paho

I'm not sure if the owners of this repository published to Platform IO. 
@icraggs, if that's the case, there seems to be a way to [claim ownership](https://github.com/platformio/platformio-registry/issues/new?labels=namespace&template=claim-ownership.md&title=Claiming+namespace+[name]) for the Paho project.

icraggs, 2023-11-29:
@CIPop no I hadn't heard of it until now. I guess the best way to help is to develop and release a new version of this library an then see if can be updated on PlatformIO.