eclipse-paho/paho.mqtt.embedded-c #256: fatal error: MQTTPacket.h: No such file or directory

Smartich0ke, 2024-01-30:
I'm trying to use this library in a Raspberry Pi Pico project but I am having issues importing it.

Currently I have the following CMakeLists file for my project:

```cmake
cmake_minimum_required(VERSION 3.13)

include(pico_sdk_import.cmake)

set(PROGRAM_NAME openclimate-basestation)
project(${PROGRAM_NAME} C CXX ASM)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(PICO_BOARD pico_w)
pico_sdk_init()

add_executable(${PROGRAM_NAME} main.cpp)

target_compile_definitions(${PROGRAM_NAME} PRIVATE
        ENV_WIFI_SSID=\"${ENV_WIFI_SSID}\"
        ENV_WIFI_PASSWORD=\"${ENV_WIFI_PASSWORD}\"
        )

# add the paho client and packet libs
add_subdirectory(${CMAKE_CURRENT_SOURCE_DIR}/libs/paho.mqtt.embedded-c/MQTTClient)
add_subdirectory(${CMAKE_CURRENT_SOURCE_DIR}/libs/paho.mqtt.embedded-c/MQTTPacket)

# include the paho client and packet headers
target_include_directories(${PROGRAM_NAME} PRIVATE
    ${CMAKE_CURRENT_SOURCE_DIR}/libs/paho.mqtt.embedded-c/MQTTClient/src
    ${CMAKE_CURRENT_SOURCE_DIR}/libs/paho.mqtt.embedded-c/MQTTPacket/src
)

add_subdirectory(${PICO_SDK_PATH}/lib/lwip lwip)
target_include_directories(${PROGRAM_NAME} PRIVATE ${PICO_SDK_PATH}/lib/lwip/include)

target_include_directories(${PROGRAM_NAME} PRIVATE ${CMAKE_CURRENT_SOURCE_DIR})

target_link_libraries(${PROGRAM_NAME}
    pico_stdlib
    hardware_i2c
    pico_cyw43_arch_lwip_threadsafe_background
    # link paho client and packet libs
    paho-mqttpacket
    paho-mqttclient
)

pico_enable_stdio_usb(${PROGRAM_NAME} 1)
pico_enable_stdio_uart(${PROGRAM_NAME} 1)
pico_add_extra_outputs(${PROGRAM_NAME})
```

But upon compiling, it looks like it cant find the library names I specified:
```
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: cannot find -lpaho-mqttpacket: No such file or directory
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: cannot find -lpaho-mqttclient: No such file or directory
```

I have double checked that the names match what is thep orject names for both libraries in their respective CMakeLists files.

I have made sure the paths are correct too. I have my project dir alongside my pico-sdk dir with this repo under the 'libs/' dir of my project ad seen in my cmake file above.

MickeWest, 2024-07-04:
Hi -
I am running into similar issues. Did you ever get this resolved?

Thanks,

  Michael

Smartich0ke, 2024-07-04:
No, unfortunately. I ended up using the built in MQTT client in LWIP.

crispycookies, 2024-11-26:
@Smartich0ke `MQTTPacket` is called `paho-embed-mqtt3c`. No idea about the C++ client though.