eclipse-paho/paho.mqtt.embedded-c #267: Result of a code analysis

DeJusten, 2025-01-22:
In my research project, I analyzed the MQTT stack for buffer overflow errors. The analysis is based on a code review. In particular, it was checked whether all pointers point to a valid area. I found no errors regarding a buffer overflow, so attackers cannot use the protocol as a gateway into a system.
However, there are errors in the code, e.g.
- Errors when sending data when it cannot be sent in one go
- Sloppy handling of error return values ​​(when setting the return values ​​and when checking them)
- Determination of the packet size consisting of FixedHeader+VriablenHeader+Payload is incorrect, which leads to a 'drop' of an actually 'suitable' frame
- Filter condition is not SPEC compliant
- Sloppy handling of data type size_t, which leads to problems in systems with sizeof(int)!=sizeof(size_t)
- Sloppy handling of Publish&Co with QoS > 0 (e.g. MessageID is not checked)
- Some code optimizations

Unfortunately, I don't have the time to correct these errors. However, if you are interested, I would be happy to send you a detailed description here.

NeroReflex, 2025-02-01:
I am interested in pretty much all of them. The sloppy return errors code handling is something I have spotted too. I plan to enhance the code over time... One PR at a time.

DeJusten, 2025-02-03:
The error description is attached.
I have noted possible solutions for the 'simple' errors. Since I don't have a test environment, I find it difficult to commit them myself.
Two errors require a few more code changes (global static variable, missing error check when calling MQTTPacket_decode())
If QoS>0, the acknowledged PacketID should be compared with the sent PacketID. These changes (especially if QoS=2) require more effort. Instead of fixing the error, the problem could also be named in the foreword.

If you have any questions, please ask

For the sake of simplicity, I have attached all routines with errors. I have added the errors themselves as a comment with my initials DJ. I hope it is OK that the error description is written in German

/*******************************************************************************
  MQTTUnsubscribeClient.c
 *******************************************************************************/

int MQTTDeserialize_unsuback(unsigned short* packetid, unsigned char* buf, int32_t buflen)
{
	unsigned char type = 0;
	unsigned char dup = 0;
	int rc = 0;

	FUNC_ENTRY;
	rc = MQTTDeserialize_ack(&type, &dup, packetid, buf, buflen);
//DJ rc sollte auf Gültigkeit geprüft werden! Momentan ist der Rückgabewert
//   von MQTTDeserialize_unsuback() auch dann 1, 
//   wenn MQTTDeserialize_ack() als gültig erkannt wird
//   if(MQTTDeserialize_ack(..) == 1) && (type == UNSUBACK)) {rc=1;}
	if (type == UNSUBACK) {
		rc = 1;
    } 
    else {
    }

	FUNC_EXIT_RC(rc);
	return rc;
}

/*******************************************************************************
  MQTTConnectClient.c
 *******************************************************************************/
int MQTTSerialize_connectLength(MQTTPacket_connectData* options)
{
	int32_t len = 0;

	FUNC_ENTRY;

	if (options->MQTTVersion == 3)
		len = 12; /* variable depending on MQTT or MQIsdp */
	else if (options->MQTTVersion == 4)
		len = 10; 
//DJ Wenn MQTTVersion ungleich 3 und 4 wird len nicht gesetzt!	
//   Am einfachsten else if()  wie bei MQTTSerialize_connect() entfernen
//   oder in MQTTSerialize_connect() else if() wie hier einfügen
//   und über ein weiteres else einen Fehler an den Aufrufer melden

	len += MQTTstrlen(options->clientID)+2;
	if (options->willFlag)
		len += MQTTstrlen(options->will.topicName)+2 + MQTTstrlen(options->will.message)+2;
	if (options->username.cstring || options->username.lenstring.data)
		len += MQTTstrlen(options->username)+2;
	if (options->password.cstring || options->password.lenstring.data)
		len += MQTTstrlen(options->password)+2;

	FUNC_EXIT_RC(len);
	return len;
}

int MQTTSerialize_connect(unsigned char* buf, int32_t buflen, MQTTPacket_connectData* options)
{
	unsigned char *ptr = buf;
	MQTTHeader header = {0};
	MQTTConnectFlags flags = {0};
[... 793 lines cut ...]
    }
    else if (message->qos == QOS2)
    {
        if (waitfor(c, PUBCOMP/*, &timer*/) == PUBCOMP)
        {
            unsigned short mypacketid;
            unsigned char dup, type;
            if (MQTTDeserialize_ack(&type, &dup, &mypacketid, c->readbuf, c->readbuf_size) != 1) {
                rc = FAILURE;
            }                
//DJ: Hier sollte mindestens mypacketid gegen message->id geprüft werden!!!                
        }
        else
            rc = FAILURE;
    }

exit:
    if (rc == FAILURE)
        MQTTCloseSession(c);

#if defined(MQTT_TASK)
	  MutexUnlock(&c->mutex);
#endif
    return rc;
}

gerhil, 2025-03-12:
Great.

There is an error in Line 259 in MQTTFormat.c.