eclipse-paho/paho.mqtt.embedded-c #275: Project activity status inquiry

JoriOtega, 2025-08-20:
Hello maintainers,

I hope this message finds you well. I noticed that the last official release of paho.mqtt.embedded-c was published in 2017. Could you kindly confirm whether this project is still being actively developed and maintained, or if it has been deprecated in favor of another Eclipse Paho MQTT library?

Having clarity on the project’s maintenance status will help us make an informed decision regarding its adoption in a new embedded systems project.

Thank you very much for your contributions to the open-source community and for your time in addressing this inquiry.

lzszero, 2025-08-25:
I have the same question. @icraggs

ashesman, 2025-11-27:
I have the same question.  But I think based on the number of open issues and lack of response to this question, I think the answer is clear!